flandreas/antares #150: Strange behavior with pull-up resistor on simulation start

richardabendroth, 2021-06-03:
I have placed an improvised 'real' button within a circuit with a pull-up resistor and when starting the simulation it seems to transmit low. 
[image]

After interacting with the circuit the first time it works flawlessly. 

[image]

What am I doing wrong? 

Here is the circuit of my improvised button: 
[image]
(I know that I'm using a limitation of the simulator to detect the high impedance)

flandreas, 2021-06-04:
Seems to be a bug in the circuit bootstrap algorithms regarding the collaboration of pull resistors and signal sources within sub circuits. I will look into it..

flandreas, 2021-06-04:
It doesn't even require sub circuits. The minimal circuit producing this bug looks like this:
[image]
So, a "weak" signal doesn't get forwarded correctly on a net on start-up

flandreas, 2021-06-04:
Important issue, requires quite some changes in the simulation start-up procedures...

richardabendroth, 2021-06-05:
What I have also noticed is that Z doesn't propagate on startup: 
[image]
After opening and closing both switches, Z is successfully propagated into the subsequent net: 
[image]

flandreas, 2021-06-07:
Fixed all known scenarios. But, honestly, as this is a pretty complex topic, I'm not entirely sure if this covers all possible scenarios. We'll see..

closed 2021-06-10 by flandreas