flandreas/antares #152: Feature Suggestion: Allow inputs of sub-circuits to become presetable in the properties editor

richardabendroth, 2021-06-03:
For circuits that have a default value, it would be nice if an input can be marked as a property so the user can set this property after placing the component symbol. The value can be read from it like from any other input. If an input is used as a property, it might not be allowed to be used as a normal input in the symbol editor. The name of the property should be the name of the input.

Kitick, 2021-06-04:
So your saying to have a way to manually set an initial condition for ICs? I guess that could be a nice quality of life.

flandreas, 2021-06-04:
The properties you would set are signal values, right? So instead of adding this circuit input as pin to the symbol, and providing the value from the surrounding circuit, you would set the value as property of the subcircuit (after adding it to the surrounding circuit).

Ok, but why? To avoid the effort of adding a constant and connecting it with the pin of the subcircuit? The property value wouldn't be visible when looking at the circuit, so providing signal values from properties would add "black magic" and therefore decrease understandability, would it?

However, this suggested feature reminds me of the concept of "generic parameters" implemented by other tools. It's like generic parameters in programming languages, applied to circuit design. But those tools use it for parameters like "Bit Width" or "Light Color", not for signal values.

Kitick, 2021-06-04:
I agree it adds a layer of abstraction for the inputs, and the cursed "black magic". I was going to point out that it goes against what Antares is for, but I wasn't fully sure what you were asking and didn't want to assume.

richardabendroth, 2021-06-04:
I was just building a HEX input button that contains two buttons and a counter with 4 bit output. I just thought it could have a preset value set' to not always start from zero. I totally get what you are saying. It doesn't make too much sense for other use cases.

flandreas, 2021-06-07:
Okay I see. I think your use case calls for the "generic circuits" feature, where your subcircuit would have a generic parameter S of type "signal", and it would use a constant with value S, and when using the subcircuit in another circuit, you would set parameter S in the properties view.

richardabendroth, 2021-06-07:
Yes exactly 😊

flandreas, 2021-06-10:
Interesting example with the startup state influencing the propagation delay! Okay, I'll set the "feature" tag and change the name of the issue to "Generic parameters".

richardabendroth, 2021-06-11:
Also, my hack for a 'real' switch is a good example (I've posted the circuits in #144). The only difference between the 'real' Switch and the 'real' Button is the 'Toggle' setting of the standard Switch Component. If the 'real' Switch component could have a generic property 'Toggle' that could be forwarded to the Toggle property of the standard Switch within the component, this could be done in just one component (like the standard Switch supports both in one component).

flandreas, 2021-06-13:
Yes, good suggestion!

closed 2021-12-14 by flandreas