flandreas/antares #154: Opinion Question for Naming Enable Pins

Kitick, 2021-06-05:
So I have made some IC's that produce an output, obviously, and in some cases I need that output to be locked, until it needs to be read. That's simply an output enable signal. Well, I've also herd it called a read enable signal. I wanted to ask you if its more appropriate to name them OE for output enable or RE for read enable. And also conversely for input, either IE or WE, Input enable or write enable. Though I haven't seen an IE before.

richardabendroth, 2021-06-05:
I use OE (mostly active low) for that since most 74's do the same. For WE I'm not very opinionated but this is what CLK normally does (like in all FlipFlops that only write on the rising - or falling for active low FlipFlops - edge of the clock). But WE also sounds reasonable if you want it to be somehow 'similar' to OE.

richardabendroth, 2021-06-05:
I need to correct myself: 
- CLK is used when setting the FlipFlops is edge-triggered 
- EN (WE) used when Q follows D as long as EN is high

Kitick, 2021-06-06:
Alright, thank you that makes sense

closed 2021-06-10 by flandreas