flandreas/antares #155: Issue with connecting wires

Kitick, 2021-06-05:
So I've decided to re-build my D-Latch circuit, and I came into a problem implementing a PRE and CLR. So I was going to have the signal just connect to the NOR gate directly, because that makes sense, but I'm not able to connect the wire to it. I know this is because its an output and an output, but similar to #5, I need this functionality.

Also similar to what I asked before in #75, where I think the wires should not care about weather its an output or input, and just let the user connect it. And if per say, its an invalid circuit, then when the simulation runs and a fatal error happens, then the wire could just be highlighted red. Because from what I've found, you are allowed to do stuff like that, like connect all outputs to a single rail and output it, like in a RAM module, you connect all the byte latches together to output it.

Unless I'm missing something more fundamental?
Here are some images of what I'm trying to do, the first is the D latch with a PRE and CLR, the second is a 4 Byte RAM with a common output rail.

[image]
[image]

richardabendroth, 2021-06-05:
For the first case you are supposed to use 3-intput NOR gates instead. As @flandreas mentioned, this limitation exists to prevent beginners from falling into traps. Here is the textbook example (but as an active low version): 
[image]
For the 2nd one you need the Q's to be high impedance when OE is low. You could just replace the last AND gate with a tri-state buffer and connect OE to it directly. Then the application allows you to connect all the Q's together into a bus net. What your are building is simply a standard active high D flip flop with PRE and CLR followed by a tri-state buffer but in one component.

richardabendroth, 2021-06-05:
But I get the general idea of being able to connect anything to anything. For example, as far as I was able to try with my limited skills, it is not possible to build a Transmission Gate D Flip Flop (TGFF). It operates not only on 0 and 1 but also on Z. But having said that, allowing this would - I guess - lead to a lot of problems in the Simulation Engine. I just tried and you can build a TGFF in Logism but it won't simulate due to immediate oscillation issues. (And using Logism showed me how nice the UX of the wiring in Antares is compared to it)

Kitick, 2021-06-06:
That's true, I thought of that somewhere, just to use tri-state buffers. I guess that all makes sense, but for me I never properly learned why you cant tie outputs to outputs. Other than why you said, then it becomes an OR gate, and then outputs can backflow. However if you design the circuit properly, that should matter. But I could see that as a better practice to not. Because for the example I gave, what if you didn't want to have to use a 3-input chip, and/or, you don't want to use more NOR or NAND gates? But I guess if its a bad practice then its better to just use more/different chips I suppose.

You can actually see this with the tunnels, if you have your circuit with output ports connect to common tunnels, you get what I was saying before, it lets you, but errors out. I guess it just not letting you saves the hassle.

Also just to clarify, I am also relatively new to circuits themselves, however I already have a strong understanding of the logic and systems involved, because of what I learned from my years in computer science. But more circuit specific stuff like outputs to outputs is where I don't have as much experience.

Here's the tunnel thing
[image]

richardabendroth, 2021-06-06:
From my googling, I can tell you that the general rule is to use gates and connect two outputs to a net. However, rules without reason are worth nothing. So I read a bit into it to also understand this matter a tiny bit better. 
Regarding the common bus topic, the general recommendation seems to be the usage of tri-state buffers that contain MOSFETs in the form of a transmission gate. Those ensure that the output is effectively separated from the subsequent circuit as long as the transmission gate is not conducting. This seems to be important especially on circuit startup when one gate is already powered up and puts out a logic signal and another one is not powered up yet. This would lead to the 2nd IC to be in an indeterminate state. This [Article](https://e2e.ti.com/blogs_/b/analogwire/posts/logic-gates-and-switches-with-ioff-empowering-you-to-power-down) from TI states that even after the 2nd IC powers up, its state could still be bogus. But again, this is just superficial knowledge from googling around. 
Regarding the two outputs feeding into the same net without the net being a tri-state bus: From reading in forums it is possible to do that nevertheless but you need to add diodes and resistors to ensure the stuff mentioned in the TI article doesn't happen. Also, people say it makes debugging very hard. However, you can use ICs with open collector outputs in combination with resistors. My opinion: this means a lot of analog fiddling in a digital circuit, adding more potential error sources, and making logic-level math (e.g. resistor values) even more complicated. 
I totally get the point that 
a) space on a breadboard or a PCB is very limited and every IC additionally adds a bunch of extra wires and caps and whatnot. 
b) having exotic multi-input gates from every type and kind at hand is obviously not the most common and also not cheap. Some of those 'special' gates are also to be considered unobtanium. 
But I guess this is a topic where beginners like us here have to trust the general rules until we understand what is really behind it. Which for my side will take a while 😅 

Regarding the tunnels: yes, you can do the same using either a component with input/output pins connected together within it or use some bi-directional splitters connected together (which unfortunately you can't set to having only 1 bit iirc).

flandreas, 2021-06-24:
Regarding connecting: Feature #134 of Release 0.7.0 should bring some relief here.

Regarding tunnels: Tunnels don't yet support bi-direction, zero propagation delay behavior like inputs/outputs, splitters and sub circuits do. That's the reason why only one wire is red in the example above. I will open a separate issue for this.