flandreas/antares #159: Delay Component does not work as described

richardabendroth, 2021-06-08:
The Manual about the Delay Component describes its behavior as follows: 
'The delay element outputs the signal present at the input after an adjustable delay time at the output.'

From my understanding there are two types of how delays can be achieved: 
1. Using Gate Propagation Delay: When the input signal changes, it takes the defined amount of ns for the same signal to be present at the output. If the input signal changes back to its original state before the signal value has propagated through the gate, the output value does not change. Example: Buffer with 20ns PD; Input Signal: 0 -> 5ns -> 1 -> 5ns -> 0; Output Signal: 0 -> 20ns -> 0. 
2. Using a Delay Circuit: The output is always the value the input has been the defined amount of ns ago. This means that if the input changes while the propagation is not completed, the output will still follow the input signal with the defined delay time. Example: Mercury Delay Line; Delay 20ns; Input Signal: 0 -> 5ns -> 1 -> 5ns -> 0; Output signal: 0 ->25ns -> 1 -> 5ns -> 0; 

From the description of the manual, the Delay Component should behave like 2. but behaves like 1. There seems to be no other way of achieving this specific kind of delay (2.) within Antares.

richardabendroth, 2021-06-08:
I just found out that when setting the delays very high both work and behave the same (like 2.). But when the propagation delay is short (like 20 ms) both seem to 'swallow' the input signal and don't change the output. Maybe I'm not getting something fundamental?

richardabendroth, 2021-06-08:
Circuit to reproduce: 
[image]
When the Switch is turned on, the AND gate puts out a 20ns high pulse and the immediately subsequent LED flashes for 20ns. However, the LEDs behind the Delay and Buffer do not light up.

flandreas, 2021-06-09:
Yes, you are right. This is probably a bug. It was supposed to function according to 2), but is implemented as 1). Am i correct that you would also prefer functionality 2)?

richardabendroth, 2021-06-09:
For the Delay for sure. Regarding the Gates I don't know enough about electronics to know if 1. is how a Gate should act?

flandreas, 2021-06-09:
Okay, fixed. Regarding logic gates: I tried this in a early stage of Antares Development, but I couldn't make sequential circuits like flip-flops work correctly with 1)

richardabendroth, 2021-06-09:
I mean even if having Gates act like 1. might be not correct electronically (which I really don't know!) it offers another form of delay which is quite nice. This actually makes the gate delay a cancelable delay. Am I correct?

closed 2021-06-10 by flandreas