flandreas/antares #160: 'Internally' Name in Button not always working when Symbol is placed

richardabendroth, 2021-06-08:
I have a Symbol with just two Buttons and two outputs but no background. Sometimes, when placing the component or re-opening a circuit where the button is used, the 'Internally' Name for one of the buttons is missing. 
In situ within the circuit: 
[image]
In the Explorer Preview: 
[image]
In the Component Circuit: 
[image]
In the Component Symbol: 
[image]

After changing the 'Label Position' Property for the switch to 'Externally' and back to 'Internally' and saving will fix the problem: 
[image]

But when moving the Component into a User Folder the problem reappears and the aforementioned steps need to be performed again.

flandreas, 2021-06-09:
Okay, thank you for reporting this.

flandreas, 2021-06-09:
Couldn't reproduce the problem, but found and fixed a bug in the "controls" code where IDs of control's view and model where confused when updating controls in symbols. This could explain the problem here, but I'm not entirely sure.

I'm closing this issue; please feel free to reopen it if the problem persists after installing the new release

richardabendroth, 2021-06-09:
It's really hard to reproduce. Sometimes it happens when I change something in the circuit, sometimes not. Sometimes when Moving the Circuit into a User Folder, sometimes not. But what you are saying sounds like it could cause the problem.

closed 2021-06-10 by flandreas