flandreas/antares #162: Component Request: A nibble counter with CI/CO and async RST

richardabendroth, 2021-06-09:
For the learner, it would be very nice to also have a sync nibble (4-Bit) counter with Carry In and Out and async Reset in addition to the existing 3-Bit counter within the standard library.  
This would have the following advantages: 
- Beginners can easily earn about cascading circuits 
- 4-Bit counters are in general more useful than 3-Bit counters when going beyond understanding counter internals but actually using them within a circuit 
- Async Reset provides a way to understand why there are different FlipFlops with different functionality (in this case CLR). 

There could be even an additional nibble counter with LOAD to help understand why there are FlipFlops with PRE. But having all of this in one Circuit just seems to be too complicated for a learner. 
I have also thought about if there should be an additional Up/Down Counter but I actually think this adds little didactic value as it is just a lot of implicit conditional logic within the aforementioned counter circuits. I think for the learner a Down-only nibble counter would be of more explainable value.

Kitick, 2021-06-11:
I agree, I actually made one for myself because I needed the functionality, its a full 8 Bit counter with Master-Slave JK flip flops. It has a load feature as you said, to set the counter to a specific state.

The other things your asking, like a carry out, is actually a good idea, I want to add that to mine now actually! Though it being 8-bit already is kind of enough though. Anyway, an UP-DOWN counter is a little more complex, but totally doable. Weather it should be standard I leave that to @flandreas, but its a good idea.

richardabendroth, 2021-06-11:
That's a perfect example case where I would love to be able to easily share a counter circuit that I've built with @Kitick 😊

Kitick, 2021-06-11:
I agree, I feel like having a way to share stuff and get feedback would be really cool in the learning experience! Though I know you said that you wanted a "Shared Library" In the web version of Antares, and that's fine, but maybe the desktop could have access to the same database?

Also @richardabendroth I'd love to see it! This is the one I made if you want. The AND gates for the carry signal are a little tight than I'd like, but I don't want to have to re organize all the wires!
[image]

flandreas, 2021-06-13:
I hear you :-)

Note however that you can already now export your project as a zip file. I've just noticed though that it's not yet possible to export libraries, although they are technically (almost) equivalent to project.

So, do you currently use custom libraries? Would it be helpful if you could export and import libraries from others?

richardabendroth, 2021-06-14:
I have to admit that I don't really use the libraries. I basically have one big project file with everything in it. It would be smarter of course to use a library. But then I would really love to have a way to just export 1 component and share it with someone. The question is how to deal with dependencies then I guess.

flandreas, 2021-06-15:
I understand. Although there is already now a way to move components between libraries, you still have to take care of dependencies, and your project can only be based on a single library.
So, the library concept indeed needs improvements on the long run.

closed 2021-07-30 by flandreas