flandreas/antares #164: Explorer Side Bar Width and Circuit Area Width not correctly restored

richardabendroth, 2021-06-09:
Switch to Window Mode: 
[image]

Hide Explorer: 
[image]

Maximize Window: 
[image]

Re-open Explorer: 
[image]

flandreas, 2021-06-09:
I've seen this once a long time ago when I've built the first release on Windows.

One of the problems is that the tabs at the left and right borders ("Explorer", "Scenarios", "Issues") should be vertical instead of horizontal. Are they always horizontal in your installation, or just now, when the re-layout problem occurs? I've also seen these horizontal tabs once on Windows after I switched to another theme.

richardabendroth, 2021-06-09:
I think always: 

[image]
[image]

closed 2021-09-07 by flandreas