flandreas/antares #166: More RGB LED Suggestions

Kitick, 2021-06-10:
The LEDs are perfect side for a few things

1: Can there be a way to remove the border, or make it the same color as the LED, as was mentioned in #145?

2: A way to make the LEDs smaller, so large arrays and grids can be shown with less space.

3: The 000000 state should be fully black, instead of grey on first power up.

4: This ones more complex, a system to have RGB LEDs have a location ID, so that when you copy paste them to make larger arrays, you don't have to plug a bunch of wires. It was more in line with the tunnels, so a tunnel would be named, 0,0 for example, and then copying 0,0 and pasting, enumerates the name to 0,1 (y,x)

4.5: Another Idea, a way to have the LEDs be mass made in a grid, so like, you can say, import a 8x8 grid, and each LED has its 24 bit input tied to a tunnel, and the tunnels name is its location, 0,0; 1,2, 6,7,. Weather it should be x,y or y,x im unsure about, because in computer science its x,y, but when implementing 2D arrays, you loop as y,x.

flandreas, 2021-06-13:
Thank you for your feedback.
1. Yes, I've promised you omitting the border, but I wanted to make the release as soon as possible.
2. Yes, that makes sense.
3. Okay. It currently uses the "dark light color aspect", which is never entirely black. But in case of the RGB LED, which indeed produces entirely dark when receiving 0x, the black color makes also sense at start-up.
4. First of all, this is kind of the purpose of the "LED Matrix" component. But since that one requires multiplexed operation, it probably doesn't suit your needs.
4.5. This is probably beyond the typical application of LEDs. I think your are asking for an "VGA Display" component, right?

Kitick, 2021-06-13:
Ah alright thanks!

And I guess yeah, if there could just be a VGA display component that would actually solve my problem entirely. Where its a display where you could enter the screen size, and you input the color data and it draws it left to right top to bottom, as a real display does. Though doing it with RGB LEDs is fun, that would be a lot easier for more complex displays!

Ill actually open a new issue with that suggestion

flandreas, 2021-06-24:
Implemented 1, 2 and 3.

closed 2021-06-24 by flandreas