flandreas/antares #168: Represent Clock Delays as the Frequency

Kitick, 2021-06-11:
Having the clocks input be the period of the clock is a handy feature, but I want to add a way to represent it as 'Hz', or 'MHz' in this case with 'us'. Now yes, you can just take the inverse of the period and that's the frequency, and vice versa, but I still think having both would be a cool feature.

And add a way to change the units, so you can set periods to be either 's', 'ms', 'us', or 'ns', (Mabey 'ps'?) and frequency's to be 'Hz', 'KHz', 'MHz', and 'GHz' respectively. (Some of those use Greek symbols, I'm just lazy to find the ascii for them :p)

Kitick, 2021-06-11:
I just realized this is similar to #157, but I my point still stands.

flandreas, 2021-06-13:
Yes, good suggestion.

closed 2021-07-02 by flandreas