flandreas/antares #169: Component suggestion

Kitick, 2021-06-11:
I actually don't know what they're called, but its a switch, that has three pins, A, B and Y (or Q?). And whatever the switch is set to, it passes that signal to Y. So if the switch is in A, then whatever A is, Y is, if its on B, then if its B, then B. The layout is actually A Y B, but same difference.

I have a picture of it if it helps:
[image]

richardabendroth, 2021-06-11:
It's called double throw switch. I can build one for you based on the one I've posted in #144.

Kitick, 2021-06-11:
Ah, I knew that it had 'Throw' in it I just was unsure exactly. Thanks for clarifying!

I think you can just use a 2:1 Multiplexor, unless the output should carry undefined when A or B is undefined.

richardabendroth, 2021-06-11:
Well yes and no. If you indeed don't care about the Z signal, the normal switch also works using an AND gate. But especially for the Double Throw Switch, which does not only direct the signal into the selected direction, it does also allow you to combine signals from two sources and exclusively forward the selected one to the receiving net. So the 'Y' goes both ways: A or B to Y and Y to A or B.

richardabendroth, 2021-06-11:
This is what I came up with: 
[image]
[image]
[image]

richardabendroth, 2021-06-11:
But indeed, having this as a standard component with the right button symbol and without my direction 'DISPLAY' hack would be wonderful!

richardabendroth, 2021-06-11:
UPDATE: Above circuit seems to glitch in Y to A or B condition. I'm investigating...

Kitick, 2021-06-12:
What's stopping you from just doing this?

[image]

Kitick, 2021-06-12:
UPDATE:

Oh I see, your having it be two way, ahh, alright I get it, that seams like it would need more logic, lemme give it a try and see what I come up with

flandreas, 2021-06-19:
@richardabendroth I've reproduced your circuit for analysis, and it seems to just work as expected, at least on my development version 0.6.2-beta. Could you please explain what you meant by "seems to glitch in Y to A or B"?

flandreas, 2022-02-15:
Implemented with release 0.20.0

closed 2022-02-15 by flandreas