flandreas/antares #170: Tri-State Buffer wire glitching when changing Control Input Direction while Signal connected

richardabendroth, 2021-06-11:
The Signal wire of the Tri-State Buffer glitches when changing Control Input Direction while a signal net is connected. 
[image]

Steps to reproduce: 
1. Place a Tri-State Buffer
2. Connect any wire 
3. Change 'Control Input Signal Direction' property to 'left' 

The problem disappears after saving and reopening or when deleting and reconnecting signal wire after Control Input Direction property change.

flandreas, 2021-06-13:
Thank you for reporting this.

flandreas, 2021-06-14:
The current version rebuilt the component's view from scratch when the "signal direction" property has been changed, thereby loosing state information. The fix keeps and adjusts the existing view. Unit test added.

closed 2021-06-24 by flandreas