flandreas/antares #172: Error when Starting Simulation

richardabendroth, 2021-06-11:
```
Version: 0.6.1
Move component 'LED' with ID 96 by mouse
Change property 'Orientation' of component 96 to 'North'
Change property 'Orientation' of component 96 to 'South'
Start creating new EdgeView at PortView
Connect output of Double Throw Switch with input of LED
Move component 'LED' with ID 96 by mouse
Move component 'LED' with ID 96 by mouse
Copy component 'Power' with ID 10
Preparing paste from clipboard
Paste component 'Power' with ID 98
Move component 'Power' with ID 98 by mouse
Start creating new EdgeView at PortView
Connect output of Power with input of Double Throw Switch
Copy component 'Ground' with ID 70
Preparing paste from clipboard
Paste component 'Ground' with ID 100
Move component 'Ground' with ID 100 by mouse
Start creating new EdgeView at PortView
Connect output of Ground with input of Double Throw Switch
Enter execution mode (deep = true)
java.lang.ClassCastException: class ch.scorpion.antares.model.g.r cannot be cast to class ch.scorpion.jabbah.graph.model.A (ch.scorpion.antares.model.g.r and ch.scorpion.jabbah.graph.model.A are in unnamed module of loader 'app')
	at ch.scorpion.jabbah.graph.model.d.f.resolve(SourceFile:148)
	at ch.scorpion.jabbah.e.j.a(SourceFile:71)
	at ch.scorpion.jabbah.e.j.a(SourceFile:60)
	at ch.scorpion.jabbah.e.t.a(SourceFile:63)
	at ch.scorpion.jabbah.e.o.a(SourceFile:31)
	at ch.scorpion.jabbah.e.a.a(SourceFile:43)
	at ch.scorpion.jabbah.e.a.a(SourceFile:28)
	at ch.scorpion.jabbah.graph.m.a(SourceFile:228)
	at ch.scorpion.jabbah.graph.model.f.r.a(SourceFile:226)
	at ch.scorpion.jabbah.graph.model.f.r.a(SourceFile:212)
	at ch.scorpion.jabbah.graph.model.b.a.a(SourceFile:147)
	at ch.scorpion.jabbah.graph.view.N.a(SourceFile:1093)
	at ch.scorpion.jabbah.graph.view.P.invoke(SourceFile:1045)
	at ch.scorpion.jabbah.base.b.a.b(SourceFile:73)
	at ch.scorpion.jabbah.d.e.o.b(SourceFile:1344)
	at ch.scorpion.jabbah.graph.a.k.invoke(SourceFile:1089)
	at ch.scorpion.jabbah.base.Q.run(SourceFile:22)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

```

flandreas, 2021-06-13:
Sadly, a zombie bug. One moment you think you've killed them, and the next moment, the rise from their tomb and start hunting you again..

richardabendroth, 2021-06-14:
I know those too well 😅

closed 2021-06-24 by flandreas