flandreas/antares #175: LED is outputting 1?

richardabendroth, 2021-06-11:
In the following circuit, the net on B and the net on Y are both 1 but B is only connected to an LED and both tri-state buffers that isolate its net within the circuit are non-conducting. What is happening? Shouldn't turn B into Z state?
[image]
[image]

flandreas, 2021-06-13:
Yes, there is something going wrong. Could you please attach your project for analysis?

richardabendroth, 2021-06-14:
Oh god, all my learning circuits are in a monster project. I'll try to figure out a good way to get it out of there.

flandreas, 2021-06-15:
Never mind, I think I can reproduce your scenario on my own.

flandreas, 2021-06-19:
See also #169

flandreas, 2021-06-19:
@richardabendroth I've tried to reproduce your circuit, but wasn't able to figure out how the circuit looks like in the regions missing in your screenshot. May I ask to to post a screenshot containing the entire circuit?

flandreas, 2021-10-01:
Probably related to #227. If not, please reopen.

closed 2021-10-01 by flandreas