flandreas/antares #176: Component Suggestion, VGA Display

Kitick, 2021-06-13:
As was talked about in #166, Have a component that's a VGA display where you can enter a screen size, and feed it color data and it will draw the colors from left to right top to bottom. And with the LED matrix, it doesn't work for what I want to do, because I wanted to use colors and have it be larger than what it is.

It would be like the terminal component actually, except this one is for colors instead of characters

flandreas, 2021-06-14:
Initiated by #166

flandreas, 2021-12-29:
Implemented as "Video RAM" component. That's not exactly the same as a "VGA Display" component, but serves the purposes of displaying arbitrary images. I hope that's sufficient for the moment.

closed 2021-12-29 by flandreas