flandreas/antares #177: NAND Gate transistor layout can display wrong value

Kitick, 2021-06-14:
When connecting transistors up in a NAND gate configuration, and powering on A, pulls the value to 0, even though B is still outputting a value. And if you enable B then disable, it then works.

Here's some images:
[image]
[image]
[image]

flandreas, 2021-06-14:
Note that transistors as they are currently implemented in Antares are just another representation of controlled buffers. Therefore, they are not bi-directional.

I'm not an expert in transistor-based circuits, but 'd say that what you are trying to do here is RTL logic, which I don't think will work with the current implementation, right? That's also why the example project "Logic Gates from transistors" use TTL logic.

But please correct me if I'm wrong.

richardabendroth, 2021-06-14:
I think this is what @flandreas means: 
[image]

My question would be: Shouldn't the FET have a body diode?

Kitick, 2021-06-14:
Ah, that makes sense, I've heard of RTL and TTL logic based transistors, however I'm not the best at them either.
What I think the problem is though is Antares makes the 0 flow though, and then it forgets the the other transistor is outputting a 1. And only when you re-assert B does it remember.

closed 2021-06-24 by flandreas