flandreas/antares #179: Transistor Question

Kitick, 2021-06-14:
So I've learned that the way to build an AND gate is to tie two N-Type transistors in series and tie the inputs of the gate to the Base of each transistor. And this is all well and good, but as I was playing around, I noticed an odd optimization, you can omit one of the transistors because you can treat it like a Data and an Enable signal, however, it effectively does the same operation. And this is actually how the tri state buffer works, its essentially an AND gate. Just with an Undefined state.

So I wanted to ask here why AND gates specifically are designed with two transistors instead of one, wouldn't using less transistors in the die of the chip make it easier to shove more in the same space and use less power and heat at the same time? Or am I missing something more fundamental that isn't shown with digital signals?

Here's a image of what I'm talking about, above is a standard AND gate, and below is my Data-Enable.
[image]

richardabendroth, 2021-06-14:
As I understand it, the Mnemonics are just a simplification. 

An AND Gate has only two defined output values (0 or 1) which are defined by a function of A and B. Therefore, you even need more than two transistors that, when both bases are high, connect the output to VCC and otherwise connect the output to ground. Since electricity flows from high to low potential and FETs have a body diode, you even need two different kinds of transistors in an AND gate to be able to have current flowing either out of or into to the output.  Furthermore, each gate is also a buffer since the output is not connected to the input but to VCC or GND depending on the inputs. 
Simple example using transistors: 
[image]
'Real' Example in CMOS:
[image] 

For the Tri-State Buffer you got essentially three possible output values (0, 1 or Z). 0 and 1 are just following the input as long as EN is high. Otherwise, the output is high impedance (or floating), practically isolating VCC or GND from the output. The isolation part of a tri-state buffer also consists of two FETs since it needs to allow current to flow in both directions depending on if the signal is connected to VCC or GND.  Tri-State buffers are made from Transmission Gates which look like this: 
[image]
Since they are called Tri-State buffers, they also contain a buffer that follows the value of the input. Additionally, there is said transmission gate in between the buffer and the output. 
Also, Buffers contain transistors as the output is not directly connected to the input even though the Mnemonics shows a straight line: 
[image]

As I understand it, Mnemonics are there to illustrate the simplest, functionally equivalent circuit. In reality, this would never work since the logic level needs to be regenerated after each gate to allow the construction of circuits that have millions of gates in sequence.

Kitick, 2021-06-14:
Wow that's a lot more explanation then I thought thank you!
I'm still a little confused, so your saying you actually need more transistors to make an AND gate? Its just simplified with these kinds of digital transistors?

And alright so, I understand what you mean by reconstruction, but isn't that done as well with just the single transistor or no?

And I see, a Tri-state buffer is completely different logic, because I noticed that you could make a tri state buffer replica, if you just used my data enable transistor layout, but take out the pull resister. Then it works the same, at least digitally.

Kitick, 2021-06-14:
Also I know in the other discussion #177 there was talk about RTL and TTL based transistors, well I've also heard about CMOS transistor logic as you said in your explanation, what's the difference between them all? Google isn't helping too much because they all seam to do the same thing, but differently?

closed 2021-06-24 by flandreas