flandreas/antares #180: Discussion: Should floating inputs be Z or 0?

richardabendroth, 2021-06-14:
Currently, when an Input is unconnected (floating), the subcircuit receives 0. Shouldn't this be Z instead? As far as I understand, the simulation engine anyways assumes Z as being a 0 when asserting the signal within the gates. What do you think?

Kitick, 2021-06-14:
I think Z makes more sense, because its understood that you want to tie unused outputs to either ground or power, so that it doesn't have an unstable value.
Actually, I think it would be more fun if it randomly switched gates, because technically isn't that would it do? Because the transistor would not have a full 1 or 0? When I connected a circuit with breadboards and I left an AND gate unconnected, the LED output just flickered. Until I either grounded or powered it.

flandreas, 2021-06-15:
Yes, that's an interesting question. The default is currently 0 for reasons of convenience. Some other tools do it also that way, while other tools do it as @Kitick suggests. Maybe it could be an option in the "beginner" versus "expert" realm...?

Kitick, 2021-06-15:
I think that would be a cool idea for a beginner expert idea, where beginner is 0, and expert is Z.
But would a Z signal randomly change a gate that it's connected to always, or only in expert mode?

richardabendroth, 2021-06-15:
I think it should always be Z, regardless of the mode. But as you two say, in beginners mode it would always coerce to 0 (like it does now for the exception of unary gates) and in experts mode, it could wiggle around randomly (maybe even - just kidding - when moving the mouse over it like when touching floating inputs on a real circuit 😅)

flandreas, 2021-06-24:
You're kidding, but I actually know a tool that does produce random values in this case by intention.

flandreas, 2021-07-06:
Similar to #187

closed 2021-07-15 by flandreas