flandreas/antares #181: Connection Sugesstion

Kitick, 2021-06-14:
I don't know what to name this, but another way that could make connecting stuff better is if you drag a components pin over another components pins, they connect. Kind of how wires with floating junctions auto-connect, but it would be pins to pins.

Heres an example:
[image]
[image]
[image]

richardabendroth, 2021-06-15:
Yeah, I also sometimes get annoyed when stuff is too tight together and I need to make room to fit a tiny wire in between. But: 
- What would happen if you connect them pin-to-pin and then pull the components apart? Would there be some sort of auto-generated wire?
- How can detach both components without deleting one of them, since there is no wire in between that you can delete?

Kitick, 2021-06-15:
I was assuming that it would just create a wire between them for you, so that lets say, you could drag a component in, connect it, then drag it to were it actually goes. I just feel like it is a quality of life, because I found having to place it down, connect it, and move it closer was annoying after a while.

richardabendroth, 2021-06-15:
Thinking about it a bit longer: Remembering from my use of Antares, most of this inconvenience comes from negating inputs on gates using inverters. When #113 is rolled out, the cases where this is a problem should vastly decrease. What do you think?

Kitick, 2021-06-15:
I feel that I shouldn't create to many problems, as long as its implemented carefully. And then if this idea gets put in it should just be the same thing as making a wire.

flandreas, 2021-06-24:
Yes, that's an idea I also was thinking about some time ago. It would perhaps be useful not only in the described scenario above, where you have to move parts away to gain space for connecting, but also when dragging new components from the library. But when designing such a feature, you would have to be careful not to create unwanted connection when the user just drags parts around.

A possible usecase scenario could look like this:
- Move a component around
- If matching pins of two components are located at the same location:
-- Display a auto-connection icon, something like a ticking bomb
-- If the user wants to connect these two pins, he has to wait until the ticking bomb has timed out. After that, the pins are auto-connected.
-- If the user doesn't want to connect, he has to drag the part away in time
-- And he should always be able to abort auto-connect e.g. by using ESC

Not exactly the easiest feature to implement or to use, but feasible.

Kitick, 2021-06-28:
@flandreas Interesting, that could be a good idea, but I thought it would just be, if you drag two valid pins over each other, it would highlight a bubble like in the example, and only if you drop the component does it connect. And then you can simply drag from the library, line it up, drop it, then move it again. It could also snap to the pins if it's close enough, so that lineing it up is easier and faster.

Though I totally see the problem of unwanted connections, but I feel like most circuits don't have components shoved together that tight where that would be a problem. And if they are, I feel like they just need to space their circuit more lol.

flandreas, 2021-07-02:
@Kitick Hm, yes, your suggestion is definitely simpler and therefore preferable. I was just looking for a way to avoid the intermediate "drop" before you can drag the component to its final location, but when I think about it now, my "ticking bomb" approach is probably too confusing.

Kitick, 2021-07-19:
@flandreas I agree, though your idea was good, my thing was that I dont always like timed things, because then you have to just wait the few seconds for it to trigger, and by that time it may feel like your wasting time rather than saving it

But I also agree with trying to remove the intermittent 'drop', but I'm not sure of any other ways to go about it. PLus, at least in my head, you'd get fast at it, drag, drop, drag. I acually got that idea from a game I played, called Silicon Zeros, its a simplified circuit program with the goal of building a mock arithmetic cpu, and one way to fast connect pins was to place them next to each other.

flandreas, 2021-07-24:
Implemented. It's particullary useful when connecting splitters. I've also changed to connection point highlight color to green (the default "snap" color) so that it can be better distinguished from selections (orange).

https://user-images.githubusercontent.com/24633141/126862279-06af9874-d9e9-474c-bb31-cd6d7e035ac8.mov

closed 2021-07-30 by flandreas