flandreas/antares #182: Bit Width Expansion

Kitick, 2021-06-14:
Can the max value of the bit width be extended to 64 Bit? I'm currently making a new 64-bit model of my computer, and I only now realized that the max width possible is 32. And so I'd have to have two of everything to get to 64.

flandreas, 2021-06-24:
Hmm, really? I would probably be straight-forward on the model layer, because the basic data type LONG supports 2 to the 64 as max value, but I fear lots of effort in all the UI components that are supposed to display these values, such as input/output etc.

Kitick, 2021-06-28:
That's what I was thinking, is that 64 bit integers are the standard for most languages, JavaScript used 64 bit signed, and Java and C++ have longs and doubles that go to 64 bit as well.

For the inputs and outputs though, you have it able to display 32 bit as binary, mabey 64 could be hex representation only?

flandreas, 2021-07-05:
Okay, request accepted :-)

closed 2021-07-15 by flandreas