flandreas/antares #183: Decimal Representation

Kitick, 2021-06-14:
Would it be a good idea to add a decimal representation to the inputs and outputs? I'm working on a 64-Bit adder, and I'm noticing how googling the hex digits for numbers this big can get repetitive!

richardabendroth, 2021-06-17:
I think this would be another case for the Script-Only components. Building this with a circuit is quite crazy even for 16 Bit BIN2BCD: 
[image]

flandreas, 2021-06-24:
@richardabendroth It is possible that you wanted to post on issue #129?

flandreas, 2021-06-24:
@Kitick I understand your requirement, and it makes perfectly sense. I only fear that I won't be easy to implement because decimal digits don't fall nicely into the 4 digit nibble world of binary and hex representation, so the input/output views would require a new "decimal-based geometry"

Kitick, 2021-06-26:
@flandreas Yes that's why I'm not entirely sure about if it's a good idea, but here's my thought:

Like if you set a 1 or 2 bit Input to HEX, since a hex digit can go father than two bits, the input simply dosnt let you enter in higher numbers. I was thinking that the decimal representation could do the same thing, you type in the number and it has a scripted limit of it's max value. So the user can only enter numbers that will work in whatever bit width the component is.

And I think this could be a expert mode mabey? Because beginners should learn to build with binary and hex because it's more appropriate for this type of logic, and decimal can be for advanced users that know what to do but want a more easy to understand way to represent the numbers. Like with a 64 bit adder to test it!

flandreas, 2021-06-28:
@Kitick Entering data is not the problem. The problem is calculating the geometry for displaying those components also with decimal representation. Changing between binary and hexadecimal is easy because everything is a multiple of 4, at least with the available bit width. This would not be the case anymore with decimal representation.

It's not impossible though, but it's not straight forward, and I would like to do it only if there is really a need for it.

Kitick, 2021-06-28:
@flandreas Ahh I see now sorry, I didn't fully understand what you meant was the problem.

So it's more of the data structure itself with it not being inline with the nice world of 4 bit hex digits. And displaying it on the components.

Alright, I guess I just need more information, do you have examples on what parts would break? Because in my head, the representation is just that, the representation, not the value itself. I was thinking that the dats is always in binary, (or hex) and it's just displayed as a different value depending on what is set.

Oh I see, I think at least, is it more of the physical part of displaying it, or something else? I'm just trying to understand sorry for

Kitick, 2021-06-28:
(didn't mean to close it I accidentally clicked it instead of scrolling)

flandreas, 2021-07-05:
I've just finished the implementation, and it was surprisingly painless. The things that required the most attention were:

- Build group of 3 digits instead of 4 as for hex and binary
- Don't allow the user to enter certain digits in certain situation. For example, if you have 8 bit in decimal representation (max 255), you can't allow the user to enter 6 as last digit if the resulting number would be 256. This is not the case for hex and binary.

It also turned out that it's now difficult to recognize the representation in which a number is displayed. Hence the new issue #205.

closed 2021-07-15 by flandreas