flandreas/antares #184: Folders Duplicating and Corrupting Circuits

Kitick, 2021-06-15:
So, A little frustrating note, I made a 64-Bit Adder Circuit that was just 8x8 Bit adders where each one of those was 8x1 Bit adders.

And well, when I finished, I wanted to put them into a folder to organize the three, the 1 8 and 64. And I did, and I went to reorder them from 1 to 8 to 64, because 1 was at the bottom. And when I did that it asked me to move it, and I said yes, then it duplicated the 1 Bit. And after that, I was just like, oh that's fine Ill just remove the dupe. And well, I did, then I went to view the 1 Bit, and it said Error Loading Circuit. And when I went to the others, they were broken as well.

And I'm also only realizing now that I could've maybe put my project here and maybe you could have looked at it, but I deleted them out of frustration. So my bad on that part

Anyway Ill have to rebuild them, which is fine, it just was like an hours worth of work getting all the wires to look nice and connecting everything. I just wanted to post here thats a major problem with the folders.

richardabendroth, 2021-06-15:
I had the same problem a while ago and couldn't reproduce it. I think it might be connected to the Modals asking if you want to move it and if you want to save it. I think if you say yes to moving and yes no to saving, it might connect the same circuit file to both items in the explorer. But again, I can't reproduce it.

flandreas, 2021-06-15:
Thank you for reporting this. And sorry for the inconveniences you have to experience with these bugs.

 I' currently working hard on these "persistence consistency" bugs. I was able to manually reproduce the scenario #172 and to capture lots of detail trails. I'm now turning this material into extensive automated integration tests, and once I got them set up and consistently failing, I can work on fixing these bugs.

I currently guessing that they are all caused by a single bug right in the core of the system. I couldn't identify it yet, but I will!

Kitick, 2021-06-28:
It's alright things happen, there all rebuilt already, I was more annoyed with all the organization I did with the wires!

I'm sure this will get found and fixed, just needs some more time

closed 2021-09-07 by flandreas