flandreas/antares #185: Sometimes Bi Directional Outputs use wrong values

Kitick, 2021-06-15:
I Built a 8 Bit bus transceiver, and I noticed that sometimes the output doesn't match the input, but if I restart the simulation and then use the same numbers, it works fine. It only does it when changing between numbers. I actually saw a little z once that was interesting.

flandreas, 2021-06-19:
@Kitick The little z means that the signal is partially undefined. Fully undefined multi-bit signals are denoted with a large "Z".

May I ask you to post a screenshot of your circuit, or even better a project that contains the circuit in question?

Kitick, 2021-06-28:
Yes I can, however I've been out of town the past while, (why I haven't been responding very frequently) but I get back later on the 29th, so I can add one then.

Kitick, 2021-07-17:
Hello, its been a bit, regarding this, I think 1.9 fixed this, because I haven't encountered it yet, but I'm still testing, ill update this if it occurs and add a screenshot

flandreas, 2021-10-01:
Same as #227

closed 2021-10-01 by flandreas