flandreas/antares #187: Unary gates produce Z output on Z input

richardabendroth, 2021-06-15:
I have to say that I am (ab?)using this behavior for detecting the Z state of a net so I'm not too eager to get this fixed but I have noticed that if you feed Z into the input of a unary gate (Buffer, Inverter, Tri-State-Buffer) the output is Z as well. I think this should be 0 (actually it should be X because it's the 2nd form of undefined). Here is an example (and also an example of how I use it to detect Z to make the 'Real Swtich' work). 

[image]

If you decide to change this, could you add a Z-Detector Component? Because it would break some of my circuits if there is no way left anymore to detect the Z state of a net.

Kitick, 2021-06-15:
Well technically you shouldn't be able to detect the Z state, because in real circuits that's a floating point value, and transistors don't know if its a 1 or 0, so it randomly just switches between them.
Unless there are real ways to detect it, but I dont know of any

richardabendroth, 2021-06-15:
Yes you are absolutely right. I just use it in my bi-directional switch hacks. I'm not using it to build real circuits, just extending the basic components with it basically. 🙈

flandreas, 2021-06-24:
In case of the inverter, it is currently implemented like in Logisim: If the input is undefined, the output is also undefined, unless the option "Gate Output when undefined" is set to "Error for undefined output".

Do you know how a real inverter would behave? Would it produce 0 if the input is undefined, as you said in your comment?

Kitick, 2021-06-28:
@flandreas It would flicker the output. As I said with my AND gate testing, I have a 74LS04 as well, and when I leave the input floating and the output to the LED it just flickers really fast. Sometimes it stays constant, but it switches or flickers at the slightest disturbance.

So I'd say if an input is Z, then the transistor should just randomly keep switching outputs. Now the question is, every tick, every few? Or just every one of it's prob delay ticks

Kitick, 2021-06-28:
I also think that outputs should randomly display a 0 or 1 on Z instead of Z. Because like was said, your not supposed to be able to detect it. Otherwise this would be trinary not binary!

But that does bring up a question, when should it display Z instead of a random 1 or 0? Or should it always be Z, and the gates output is simply a 1 or 0

flandreas, 2021-07-05:
Okay, I will probably introduce a new system setting to choose the desired behavior.

closed 2021-07-15 by flandreas