flandreas/antares #201: A more user friendly scripting interface

Kitick, 2021-06-26:
As it is the last time I looked, if you wanted to set a script for a component, you have to enter the java code that works the logic directly. Now for me that's alright, because I just looked around a bit, saw what did what, then started using the api with the java I know.

However, it's a little confusing for someone who may not know how to code in java, or code at all to work with making a script that's not a pile of bad code in the end.

So my proposal is to get a fully worked scripting interface that would allow you to use sudo commands that convert into the pins values and whatnot.
So for a D latch for example, you could say:
If(pin.EN == 1){
    pin.Q = pin.D
}
else{
    pin.Q = 0
}

I know this is still very code technical, but it can be reworked to whatever else. And it also is similar to how you already do it, I just think maybe it could make things look nicer or easier to implement.

Or even just have buttons to click that allow you to auto input the code, so there's a read pin button, that then has a select menu for the pins in the component, and then you click one, and it auto completes the code to set that pin to the value.

What do you think?

flandreas, 2021-06-28:
Thank you for your suggestion.

First of all, it's actually JavaScript, not Java.  But the syntax is very similar.

In case you didn't notice, there is an entire chapter in the user manual describing the scripting API. But of course, it's always useful to look around and study the scripts that already exist in the standard libraries.

Yes, the current script "language" is somehow technical and not straight forward to use. That's because the language is not parsed and interpreted; for reasons of limiting effort, the scripts are mere wrappers to existing domain logic, and therefore their expressibility is somewhat restricted.

I don't think it's worth the effort to provide a scripting environment that provides a lot of editing support like a full blown IDE, because scripting is and always will be something that only very advanced users will ever use, such as teachers or authors of interactive text books, and I would expect from those users to learn a language, even if it is a real programming language.

However, the reason why there will be another Antares scripting language in the future is really web security: There is no way to allow user of the upcoming web version of Antares to write scripts in their circuit, and let them be executed in other user's browsers. So, we must turn away from JavaScript in the future.

I currently envision two options for the future Antares scripting language. The first option is to include an interpreter for Verilog or VHDL. If there are already standards for hardware description languages, why not use these? The second option is to define a proprietary language. Both options have their pros and cons. I'm not yet sure which way to go.

Kitick, 2021-06-28:
Ohh interesting, I was under the impression it was Java, my bad!

And I totally see your point, it would probably take more effort than it's worth, so let me rephrase the suggestion: (and also I didn't mean to make a full IDE lol, that would be a lot of work, and not practical sense there are many well made ones alrighty available)

How about instead of a full on user interface for it, just some things for autofill or auto complete, just for quality of life?

Also I totally didn't think to look on the website for documentation I feel really dumb for that, sorry for that!

Last thing, so about the language itself, of course how you have it now is more in line with just coding in the real way to access the values, and honestly, I'd almost say develop a proprietary language, because that would allow you to have it be whatever you wanted, and work and feel your way. That's the pro I'd personally like, however using an already eablished standard also is good for it's reasons.

I'm not sure either to be honest!

flandreas, 2021-07-05:
Yes, it' a hard design decision, I'm not sure either what the best way to go is. But it will be one of the last things to do in the upcoming web version, so there is still plenty of time to think about it.

Kitick, 2021-07-17:
yes, i agree, I was more suggesting it on a whim, but it seams the plan you have is better in the long run

flandreas, 2021-09-13:
@Kitick I'm currently in the process of designing and implementing the new scripting language for Antares. The new design abandons JavaScript entirely and is instead based on a custom DSL (domain specific language). Similar to HDL languages like Verilog, ports and their signal values will be accessibly simply as variables.

As an example, let's look at the execution script for a half adder. The old script looks like this:

```
var a = elem.inputBit("A");
var b = elem.inputBit("B");
elem.setOutputBit("S", (a || b) && !(a && b));
elem.setOutputBit("C", a && b);
```

Using the new DSL, the script will look like this:

```
S = (A or B) and not (A and B)
C = A and B
```

Much simpler and more convenient!