flandreas/antares #205: Indicate Signal Representation in Inputs/Outputs

flandreas, 2021-07-05:
Now that #183 has been implemented, it can be hard to tell whether a number displayed in an input/output is a hexadecimal or a decimal number. Improve the input/output component by adding the configured signal representation designator (prefix, suffix, base subscript) to the view.

Kitick, 2021-07-17:
its kinda how the constants represent their values, just with a little subscript at the end

flandreas, 2021-07-19:
Exactly. The only challenge is to gain space in the crowded figures to place the subscript. And, because of to the user preference "Circuit -> Signal notation", I'm supposed to not only support subscript, but also prefix and suffix. Somehow.. :-)

Kitick, 2021-07-19:
Thats true, well you could just use B H D as data type prefixes (Binary, Hex, Decimal). so that you only need to add one character, than 2 16 10.
though stating the base is more immediately understood... Oh you could have just a prefix be 0x or 0b, depending if its binary or hex, and nothing if its decimal. Thats just for prefix though, the others i wouldnt know

At least with my circuits, the data does get fairly big, and looking at a binary representation is just, wow. Mabey you could color code? though that could be confusing, especially if we still wanted to implement wire colors, and or the colors for other things. yeah nevermind on that.

flandreas, 2021-07-21:
Yes, prefix and postfix should also be applicable, just like it already implemented for the "Constant" component.

flandreas, 2021-07-22:
That's how it's going to look like in 0.10 with "suffix" user preference.

[image]

Kitick, 2021-07-24:
@flandreas Awesome! I think that looks great, my only suggestion would be to have the letters be Capitalized, as to make differentiating a b or d easier. Also would help dyslexic people. And in my opinion looks nicer as B H D

flandreas, 2021-07-25:
@Kitick Don't you think that people would then perhaps confuse the suffix with a regular digit, like in "7BH"? But you're right, according to Wikipedia, both representations "7BH" and "7Bh" are valid and used. Maybe another user preference option?

Kitick, 2021-07-26:
@flandreas I actually thought that to, like with B and D especially, but yeah, I can see you point

What I would do is yes make it a user preference, but also make it small enough to where it's obvious that it isn't part of the data

closed 2021-07-30 by flandreas