flandreas/antares #208: simulation skips a few steps on first start up

Kitick, 2021-07-17:
When you first start a simulation on the slowest mode, and input values, it skippes a few visuals at first, then works as normal after the first data input

flandreas, 2021-07-19:
The visuals that are skipped are the bit animations flowing along the wire? Or do any circuit parts not display the correct state? And where do you enter data? You press a switch, or enter data in a circuit input?

Kitick, 2021-07-19:
Its just visual, at least as far as I can tell.
on my binary adder circuit, I start the sim on the slowest mode, and when I input some data into the circuit inputs, the whole circuit just instantly completes the execution as if it was on the fastest mode. But after that, then it works as normal

flandreas, 2021-07-21:
Okay, thanks, I'll check that.

flandreas, 2021-07-28:
@Kitick I've did some tests and I'm not able to reproduce the issue.

I've tested with the "Full Adder" circuit from the standard library and did the following:
- Slowest simulation mode, no pausing
- Option "Simulation / Deep Simulation" selected
- Option "Simulation / Animation" selected
- Start simulation -> some initial signal animations flow along the wires
- Click on circuit input "A" to change value from 0 to 1
- A "1" signal animation flows from the circuit input to the first half adder, and further signal animations take place until the calculation settles

So, you obviously tried something else. What are the "visuals" you are missing in your test? Or can you post an image or a video of your scenario?

flandreas, 2021-07-29:
@Kitick Okay, I accidentally ran into this issue by myself.

It's a side effect of the change in the current version that tries to omit unnecessary signal animations during circuit startup. Antares swallows all animations that are about to take place at simulation time = 0 (which is true at startup). If the first thing you do after startup is changing the value of a circuit input, that animation is also omitted, because circuit inputs have propagation delay 0 and therefore don't increase the simulation time.

However, this is still a bug because circuit inputs are supposed to use a switch's propagation delay when being operated manually by the user. I'll fix that.

closed 2021-07-30 by flandreas