flandreas/antares #209: Sorta wacky suggestion, what if there were sounds?

Kitick, 2021-07-19:
Now now yes I fully understand that this would have no baring of functionality, and it would take some work to put in, getting the files and timings and whatnot.
But just think about it, once all the major and important bugs and features are put in, What if there was settings to add like, little click sounds when you connect stuff, or delete something, and if there's and error, it might beep, or something? I feel like that might add a bit of depth to it. and it would be a setting so a user could turn them off if they wish
Oh i had another useless idea, ill make another suggestion for l for it

flandreas, 2021-08-03:
@Kitick I apologize that I didn't react to this suggestion. I've qualified it at that moment as "low priority" and was focusing on more pressing issues.

As far as I remember from software ergonomics lectures, you're supposed to be reluctant using sound effects in applications whose core domain is not related to sound. If your user interactions rely on essential feedback, you're always better off to provide this feedback in a visual way. If let's say it was on always clear to the user whether his attempt to connect parts was successful, you'd better show a visual animation than playing a sound. That's because you rarely see business application using sound effects to accompany user interactions.

However, thinking a bit longer about this idea, it could perhaps be interesting if we envision Antares as learning platform, where kids/students are supposed to do homework assignments, and some "gamification" elements, when cleverly applied, could improve their learning motivation.

So, I don't think that sounds are really useful for the current Antares desktop version, which is primarily used by teachers and advanced students for authoring circuits, but it might be interesting for the future web version, where homework assignments, gamification and all sorts of new and interesting aspects could come into play.

Kitick, 2021-08-03:
Yes, I totally understand that this would have been the lowest possible priority on the to-do list, I just saw of how you said, if there was some "game elements" like that it could make it more appealing to like you said younger kids who need that extra haptic feedback. I was more mentioning it because some games that I played where you are building circuits or whatever else, they usually had background music and everything you did made little sounds, it just added depth I guess for me.

But like I said, I'm not saying that we have to have sound, it doesn't really need it at all, but I think once we're at the point where we're adding features for fun instead of pressing bugs it could be something interesting to look at.

That does raise an interesting question, what is the target age group? As of right now it's definitely tailored to intermediates and professionals who already know a little bit about what they're doing. There's not too much for very beginners, or even younger people who maybe just getting into it. Maybe we could add an interactive tutorial in the future? That not only explains the basics of Antares, but general circuits as well?

Anyway I only closed it because I understood it was a low priority, and I guess I felt bad cluttering up the git page with unimportant things, sorry about that.

flandreas, 2021-08-04:
Regarding your question about the raget audience for Antares.

Unlike other tools in this domain (which I'm sure you know them all), Antares is not just the Desktop application you're familiar with. Right from the start, It has been designed and tailored to be a general framework for interactive, animatable graphs. The digital logic stuff is only one possible application, and only the tip of the iceberg.

As I've already explained in other discussions, my ultimate goal for the Antares project is to bring its technology to the web, where all sorts of applications will be possible. This could be online interactive textbooks authored by teachers, including online interactive homework assignments. This could be an integration into learning platforms like brilliant.org, which also has a focus on interactive learning elements. This could be learning games for kids. Or this could even be teaching novels in the style of "But how do it know?" (sic) by J. Clark Scott, but this time interactive and animated. Or something like an adventure game, where you have to build a computer from scratch in order to escape from a island..

The desktop application will always be important to me, like as you said, with more having the intermediate level target audience, but only the web version will ultimately unlock all the diverse and fun possibilities for which the Antares technology is better prepared that any other application in this domain I'm aware of. Or at least I hope it is :-)

So, yes, I totally agree with your idea for the interactive tutorial, but I would envision it as a web page, where text paragraphs, interactive Antares circuit block, videos, pictures of ancient computers and so forth go hand in hand.

closed 2026-08-16 by flandreas