flandreas/antares #210: Component suggestion, A sound player / Speaker

Kitick, 2021-07-19:
Alright, this came from #209, but this can be implemented without it

What if there was a component that played little 8 bit beep sounds, and the pitch was controlled by a binary input, and whatnot? And there could be a type selector to change it to other types of sounds, like piano or drums
This is purely a fun component, and would defiantly be after the more important stuff was put in first. But know knows, I think it could be fun, you could add beep codes to computers, or make a full on music player lol

flandreas, 2021-07-21:
Yes, that's a good suggesting for a fun feature.

flandreas, 2021-11-24:
Upcoming with release 0.15.0.
[image]

closed 2021-12-14 by flandreas