flandreas/antares #211: Add rate of execution to time status

Kitick, 2021-07-19:
Could there be a way to have the show time in the status bar have another setting called display execution speed?
Or even put that up by the slider where you change the mode, even where you could type in your own speed mabey?

It would be just a visual indicator where it would say how fast the simulation is running, like 100 ns/s, or 500 us/s, etc

flandreas, 2021-07-21:
I'm assuming that you are solely interested in the "net simulation time", without the time the simulation is idle because noting is changing in the circuit.

Are you aware that the clock component already gives you this information? During simulation, its tooltips shows the"Effective frequency" value. So, if your are driving a CPU circuit with a clock, you can quickly how fast Antares is in simulation the circuit - or how slow...

If this is not what you would expect, please let me know what use case you have in mind.

Kitick, 2021-07-26:
Ah alright, yeah it was, what I was going for was like it would say how fast Antares is simulating the circuit, like 10ns/s or something like that. But the clock has it, and I guess you don't need it for other general circuits without a clock

closed 2021-07-29 by flandreas