flandreas/antares #217: Component: Signal Replicator

Kitick, 2021-08-02:
This goes inline with the whole bit width to the gates, where there should be a component specifically designed to replicate a signal across different bit widths. So for example, you could have a 8 Bit AND gate or XOR gate, and use a switch with a replicator to act as the enable. The replicator should simply just produce a constant 0000 0000 or 1111 1111 based on the input.

Another idea is, just add the bit width to switches, and have the switch just output a constant 0x or 1x based on the width. This would replace the need for a separate component. Unless this separate replicator could have other uses?

flandreas, 2021-08-03:
Good suggestion! I think we'll go with the signal replicator component (or "Bit Extender", as other tools call it). We should probably make it slightly more versatile right from the start, such that it can map n:m signals.

flandreas, 2021-08-05:
Implemented.

[image]

Kitick, 2021-08-06:
Awesome! Looks great!

closed 2021-08-19 by flandreas