flandreas/antares #219: Saving resets the undo states

Kitick, 2021-08-03:
I totally understand why it's doing this, however I would like to suggest to have a way for the undue states to be saved even after the circuit gets saved, because they were instances where I did a small change just to see what it did, and then I clicked save by habit before starting the simulation, and then when I went back to undo it wouldn't, because I saved.

Maybe just have it always save I don't know the last 10 or states of the circuit in a log file? And when the entire program is closed those extra log files are deleted

flandreas, 2021-08-04:
Yes, I think this idea calls for something like a "save history" that is maintained in log files, and that allows you to revert to older versions of your project. This could be a useful feature, and it shouldn't be that hard to implement.

Kitick, 2021-08-06:
I agree, that could be very useful actually!

flandreas, 2022-03-23:
Implemented a "Save History" for each circuit including the possibility to restore an old version

flandreas, 2022-03-24:
Released with 0.23.0

closed 2022-03-24 by flandreas