flandreas/antares #224: Rethink "Dive into subcircuit" animation

flandreas, 2021-09-03:
Usability tests show that users would like to skip "dive into subcircuit" animations when simulation speed is in the fastest speed category "Use".

Users claim that when in speed category "Use", everything should be as fast as possible, including navigation between circuits and subcircuits. It is also worth mentioning that those users weren't aware of the possibility to skip dive animations by pressing META when clicking on the symbol. This raises the question of how to communicate this shortcut.

The goal of the dive animation feature was to help users keep a mental picture of the current state of the application, which is easier if state changes don't occur abruptly. Maybe the dive animation now just takes too long and should be much faster?

Kitick, 2021-10-10:
I feel like there should be a tool tip when you hover over it to say, click to open circuit, click meta to open immediately, or something like that.

flandreas, 2021-10-10:
@Kitick Chances are that especially younger people will never see the tooltip. Like my sons, trained by years of action gaming, the always move the mouse over the screen in lighting speed, and they aren't even aware that something like tooltips exists ;-)

What about a toast that shows up if preference "Show tips for beginners" is selected AND speed category is "Use" AND dive animation is performing? The toast would say something like "Did you know that you can skip dive animation using META?".

Kitick, 2021-10-19:
Sorry for the super late response, I feel like that could be a good idea, just a "hey did you know ..." could be a useful alternative to like you said people not seeing the tooltips for whatever reason lol.