flandreas/antares #227: Problem with the tri state handling in the simulation?

ZoBoRf, 2021-09-28:
I am trying to simulate a small circuit (MemoryRW) to control 4 x 4 bit memory (RAM 4x4) in _Antares 0.12.0._ 
The circuit consists of a memory control (MEM CTL) which emits the signals to write (MW) to the memory and to read from it (MR).
The memory is connected to the data bus through a bidirectional buffer register (BDPR) and 
uses a bidirectional bus driver internally (BDBD) to buffer the last value read or written.
Both of them use tri state gates to access/connect the data bus.

This "sort of" works in the simulation. On the first MW (hi/lo) the test pattern (1011) ist written to memory cell 1 correctly.
But the second MW signal writes a 0000, which is not correct.
The third is again ok, but the fourth not and so forth.
The read is working from 2nd MR (hi/lo) signal. The first MR reads 0000 which is not correct.

I guess there is a problem with the tri state handling in the simulation but I am not sure.

[image]

[Microcomputer.zip](https://github.com/flandreas/antares/files/7242134/Microcomputer.zip)

flandreas, 2021-09-28:
Thank you for reporting this. I'll start investigation right away.

flandreas, 2021-09-28:
I'm not a digital circuit design expert, but your design looks good to me. The problem seems to be in Antares.

When the data 1011 is to be stored for the second time, signal 1 at BDBD.R disables all write-TriStateBuffers in BDBD (the ones pointing east). Their outputs produce Z, which is illegally forwarded to the Dx pins of BDBD, replacing their correct values 1011 with ZZZZ. These ZZZZ then bounce to the inputs of the read-TriStateBuffers in BDBD (the ones pointing west), and these TriStateBuffers all produce 0000, which is forwarded to DOx of BDBD and then of course onward into the RAM.

Conclusion (Bug): Floating pin values (Z) should not displace valid 0/1 on a wire.

I'll try to fix this and release the fix as soon as possible.

flandreas, 2021-09-28:
I've investigated the problem a bit more and came to a new conclusion. Now I think there is a timing problem in your design.

Precondition:
- 1011 is stored in the RAM
- MW is 0
- BDBD.Dx is 1011
- The bus at BDBD.DIx is 1011 (determined by the output of the RAM)

Action: Press MW for the second time

Protocol:
- BDBD.R goes to 1
- The west-facing TriStateBuffers in BDBD get enabled. BDBD.DOx goes to 1011
- The east-facing TriStateBuffers in BDBD get disabled. Nobody in the entire circuit now applies a stable signal to the bus at BDBD.Dx. Therefore, Antares correctly applies ZZZZ to this bus.
- These ZZZZ arrive at the west-facing TriStateBuffers in BDBD, which are still enabled. Since enabled Antares TriStateBuffers produce output 0 with input Z, the bus BDBD.DOx goes to 0000.
- This 0000 travels over the bus to RAM.DIx and are eventually stored in the RAM

Conclusion: An undefined signal on the upper data bus is forwarded to the RAM (although converted to 0000 by the TriStateBuffers), while this should not happen before the data to be stored has been arrived from BDPR.

Maybe you should disable BDBD by setting A to 0 until a stable signal from BDPR has been established on the upper data bus?

ZoBoRf, 2021-09-28:
Thank you for you time to analyse the problem! You are right. I use examples from another simulation program, where the examples work. The delay and tristate simulation different in both programs but Antares seams to do a better job in my opinion.
I could fix my problem in the following way:

[image]

flandreas, 2021-09-29:
Great!

ZoBoRf, 2021-09-29:
Because reading a different cell after writing another doesn't always work, I have removed all timing components ("MemoryRW Tester 6" in the attached project) and am trying to figure out the right signal order by hand for writing and reading the memory.

My test plan is:
* write 1101 to memory cell 11
* write 1110 to memory cell 01
* read back from memory cell 11

At this point I get ZZ01 on the external data bus, what I think is an error.

Exact signal order to reproduce:

||||||||||||||||||
| --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | --- | ---|
|||Write 1101 to Memory Cell 11||||||||||||||||
|Data to be written to Mem|DB3|1||||||||||||||||
||DB2|1||||||||||||||||
||DB1|0||||||||||||||||
||DB0|1||||||||||||||||
|Addressed Memory Cell|A1||1|||||||||||||||
||A0||1|||||||||||||||
|Enable External Data Input|EE|||1|||||||||||||0|
|Data Direction: read from (1) / write to (0) External Data Bus|r||||1|||||||||||0||
|Enable BDBR|a|||||1|||||||||0|||
|Write to BDBR|w||||||1|0||||||||||
|Enable Memory|EM||||||||1|||||0||||
|Enable Internal Data Bus|EI|||||||||1|||0|||||
|Write to Memory|W||||||||||1|0||||||
|||||||||||||||||||
|||Write 1110 to Memory Cell 01||||||||||||||||
|Data to be written to Mem|DB3|1||||||||||||||||
||DB2|1||||||||||||||||
||DB1|1||||||||||||||||
||DB0|0||||||||||||||||
|Addressed Memory Cell|A1||0|||||||||||||||
||A0||1|||||||||||||||
|Enable External Data Input|EE|||1|||||||||||||0|
|Data Direction: read from (1) / write to (0) External Data Bus|r||||1|||||||||||0||
|Enable BDBR|a|||||1|||||||||0|||
|Write to BDBR|w||||||1|0||||||||||
|Enable Memory|EM||||||||1|||||0||||
|Enable Internal Data Bus|EI|||||||||1|||0|||||
|Write to Memory|W||||||||||1|0||||||
|||||||||||||||||||
|||Read from Memory Cell 11||||||||||||||||
|Data to be written to Mem|DB3|x||||||z||||||||||
||DB2|x||||||z||||||||||
||DB1|x||||||0||||||||||
||DB0|x||||||1||||||||||
|Addressed Memory Cell|A1||1|||||||||||||||
||A0||1|||||||||||||||
|Enable External Data Input|EE|||||||||||||||||
|Data Direction: read from (1) / write to (0) External Data Bus|r|||||||||||||||||
|Enable BDBR|a|||||1|||||0|||||||
|Write to BDBR|w||||||1|0||||||||||
|Enable Memory|EM|||1||||||0||||||||
|Enable Internal Data Bus|EI||||1||||0|||||||||
|Write to Memory|W|||||||||||||||||
|||||||||||||||||||

Here is  a screenshot of the situation:

[image]

[Microcomputer-20210929.zip](https://github.com/flandreas/antares/files/7251397/Microcomputer-20210929.zip)

flandreas, 2021-09-29:
I can reproduce the problem. The minimal reproducible example (MRE) looks like this:

https://user-images.githubusercontent.com/24633141/135290501-05bf6668-a177-48af-8b55-3892d54efa38.mov

Bidirectional pins, once having received a value from outside, refuse to consume a value from inside although the signal from outside has been gone to Z.

Thanks for reporting this. I'll try to fix it at soon as I can.

flandreas, 2021-09-30:
A short update. I've was able to implement a fix in the development version that makes your test scenario work, at least as far as I can tell. It is quite a change in the core of the simulation system, but it is now simpler and cleaner than before.

However, I'm now struggling to make my example circuit "Tanenbaum Microcomputer" work again. It is it now facing timing issues when accessing the memory. I need to fix them before I can release the change.

In the meantime, thank your for your patience.

flandreas, 2021-10-01:
Fixed with release 0.12.1

flandreas, 2021-10-01:
I have to reopen this issue and take a complete different approach to solving it. Not only does it create new known problems like #228, it will also not work with sub circuits that contain bi-directional ports when simulated with shallow simulation mode.

closed 2021-10-04 by flandreas