flandreas/antares #228: Signal from tri state buffer stays lo/hi

ZoBoRf, 2021-10-01:
After starting the simulation in 0.12.1 the situation is as follows:

[image]

now set EN=1

[image]

now set EN=0

[image]

The bidirectional outputs are not reset to Z.
Is this correct?

ZoBoRf, 2021-10-01:
Remark: 

The simulator Hades (https://tams.informatik.uni-hamburg.de/applets/hades/webdemos/90-tionline/06-ralu/tri.html) resets the outputs to Z, after the tristate buffers are deactivated.

[image]

flandreas, 2021-10-01:
No, that's not correct, thank you for pointing that out. There is indeed a unit test missing for this scenario.

closed 2021-10-04 by flandreas