flandreas/antares #272: Faced this issue

K2E7, 2022-01-14:
Version: 0.16.0
Loading settings from 'C:\Users\Skylor\AppData\Roaming\Antares\Antares.ini'
Loading preferences from 'C:\Users\Skylor\AppData\Roaming\Antares\Antares.pref'
Close SideBarPane 'null'
Open SideBarPane 'Explorer'
open Project 'My first project'
Open application data
Start creating new EdgeView at PortView
java.lang.NullPointerException
	at ch.scorpion.jabbah.graph.view.b.Z.e(SourceFile:2445)
	at ch.scorpion.jabbah.graph.view.b.bz.invoke(SourceFile:1187)
	at ch.scorpion.jabbah.base.j.a.a(SourceFile:192)
	at ch.scorpion.jabbah.base.j.d.a(SourceFile:80)
	at ch.scorpion.jabbah.base.j.d.b(SourceFile:1134)
	at ch.scorpion.jabbah.base.j.f.d(SourceFile:269)
	at ch.scorpion.jabbah.base.j.d.b(SourceFile:104)
	at ch.scorpion.jabbah.c.C.b(SourceFile:123)
	at ch.scorpion.jabbah.edit.g.J.c(SourceFile:197)
	at ch.scorpion.jabbah.edit.j.a.c(SourceFile:5058)
	at ch.scorpion.jabbah.edit.d.k.c(SourceFile:195)
	at ch.scorpion.jabbah.c.h.G.mouseReleased(SourceFile:324)
	at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(Unknown Source)
	at java.desktop/java.awt.Component.processMouseEvent(Unknown Source)
	at java.desktop/javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.Component.processEvent(Unknown Source)
	at java.desktop/java.awt.Container.processEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

flandreas, 2022-01-14:
Thank you for reporting this.

flandreas, 2022-01-14:
@K2E7 This is clearly a bug in Antares, but I'm still not able to reproduce it, so perhaps you can give me more information.

According to the log you've provided, you did the following:

1. You started creating a new wire by clicking on the input port of a component (without pressing ALT)
2. You dragged the mouse over another existing wire while holding the left mouse button
3. You released the left mouse button while the mouse was still over the existing wire

Can you remember something else you did? Was the green connection indicator visible over the target wire when you've released the mouse button? Did you click on other mouse buttons while you've dragged the mouse?

I do see some potential in the wiring code to make it generally more robust, but it would be good if I was able to reproduce your specific scenario.

flandreas, 2022-01-14:
@K2E7 Thank you, but I still don't understand your use case.

May I ask you to attach a screenshot of the situation from which things go wrong, or even better a small screen cast video that shows how you are trying to connect components? There should really be no need for adding components only to connect components and wires that already exist.

flandreas, 2022-01-16:
Improved robustness of connection logic regarding unexpected events from input device

closed 2022-01-20 by flandreas