flandreas/antares #282: Renaming Circuits

leonuetzel, 2022-01-24:
Feature Request: Make Renaming Circuits possible

aquamiau, 2022-01-24:
You can in the "Properties" section, but adding it to the context menu would be useful

leonuetzel, 2022-01-24:
Oh yeah, right. Didnt see that by now.
Thank you.
Im just new to this Tool, so im finding new Functions every Minute :)
Btw. im very amazed of your work. It looks really great and the simulation makes it really easy to understand the functionality of the circuit.

leonuetzel, 2022-01-24:
I just tried to rename a circuit by typing in a new Name in the "Properties" Section. It shows the new Name in the "Properties" Section but not in the Explorer Window where you can browse Folders

flandreas, 2022-01-25:
@leonuetzel The new name appears in the explorer window after you have saved the circuit.

@aquamiau Thank you for your suggestion. I will add a context menu in the explorer tree. See also issue #47.

flandreas, 2022-01-26:
Implemented new explorer context menu item for renaming circuits. Menu item will be disabled for the current circuit if unsaved changes exist.

closed 2022-02-01 by flandreas