flandreas/antares #285: Feature Request: Rotate entire selection

aquamiau, 2022-01-25:
I find myself wanting to rotate entire selections occasionally, that would save some time rearranging circuits.

aquamiau, 2022-01-25:
This also goes for mirroring

flandreas, 2022-01-25:
Interesting feature, although probably not the easiest one to implement properly, especially because wires cannot be rotated yet. I'll think about it..

flandreas, 2022-01-30:
Rotation implemented

flandreas, 2022-01-30:
Mirroring would be much more difficult, because subcircuit symbol is currently the only component that can be mirrored. Mirroring entire selections would require to make all components like gates, switches, input/outputs etc mirrorable.

Let's see whether rotation by 180 degrees can cover most of your use cases. Otherwise, please open a separate feature request for mirroring.

closed 2022-02-01 by flandreas