flandreas/antares #288: Using extra Mouse Button "previous" makes Antares crash

leonuetzel, 2022-01-25:
My Mouse got additional Buttons which execute the "forward" or "previous" Instructions.
Using "previous" (= go back) from a subcircuit to return to parent circuit makes antares crash.

flandreas, 2022-01-26:
@leonuetzel I assume by "crash" you mean Antares is showing an "Unexpected Error" dialog. May I ask you to produce the error once more, copy/paste the error log from the dialog, and attach it to this issue?

leonuetzel, 2022-01-26:
Version: 0.17.1
Change property 'Repräsentation' of component 3 to 'Binär'
Change property 'Name' of component 3 to 'R'
Change property 'Anzahl Bits' of component 3 to '1'
Copy component 'Eingang' with ID 3
Preparing paste from clipboard
Paste component 'Eingang' with ID 4
Change property 'Name' of component 4 to 'CLK'
Move component 'Eingang' with ID 4 by mouse
Open application data
Open application data
Move component 'Anschluss' with ID 4 by mouse
Move component 'Anschluss' with ID 5 by mouse
Move component 'Anschluss' with ID 4 by mouse
Move component 'Anschluss' with ID 5 by mouse
Move component 'Anschluss' with ID 5 by mouse
Change property 'Name' of component 2 to 'ADDRESS'
Change property 'Name' of component 1 to 'DATA'
Move component 'Eingang' with ID 1 by mouse
Move component 'Eingang' with ID 3 by mouse
Move component 'Eingang' with ID 4 by mouse
java.lang.IllegalArgumentException: unknown button 4
	at ch.scorpion.jabbah.c.i.H.k(SourceFile:5255)
	at ch.scorpion.jabbah.c.i.aF.b(SourceFile:100)
	at ch.scorpion.jabbah.c.i.G.mousePressed(SourceFile:328)
	at java.desktop/java.awt.AWTEventMulticaster.mousePressed(Unknown Source)
	at java.desktop/java.awt.AWTEventMulticaster.mousePressed(Unknown Source)
	at java.desktop/java.awt.AWTEventMulticaster.mousePressed(Unknown Source)
	at java.desktop/java.awt.Component.processMouseEvent(Unknown Source)
	at java.desktop/javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.Component.processEvent(Unknown Source)
	at java.desktop/java.awt.Container.processEvent(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.Component.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$5.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

flandreas, 2022-01-26:
@leonuetzel Great, thanks a lot, I see the cause of the problem.

closed 2022-02-01 by flandreas