flandreas/antares #295: Connecting a wire to another wire isnt possible

leonuetzel, 2022-02-01:
Im trying to connect two wires together that have been released already.
The Program doesnt connect them together (see Pictures
[image]
[image]
)

flandreas, 2022-02-02:
Yes, this has not yet been implemented. It's probably a special case of your issue #296.

flandreas, 2022-02-04:
Workaround: After #5 has been released, you can drag the end point onto the other wire and create a new junction, and then trim off the dangling wire manually. I'm not sure whether it's worth it to support this special case as well.

closed 2022-02-09 by flandreas