flandreas/antares #297: Exporting ROM Content and reimporting doesnt work

leonuetzel, 2022-02-02:
Exporting the ROM Content to a File works. But reimporting that exported File doesnt work.
Error Message: "The Import File cant be read."

flandreas, 2022-02-02:
@leonuetzel Thank you for reporting this. Cannot reproduce. May I ask you to attach the exported file? And what are the characteristics of your ROM (address width, data width)?

leonuetzel, 2022-02-02:
[image]
This is the ROM Content: https://www.transfernow.net/dl/20220202nUirqx1J

leonuetzel, 2022-02-02:
[image]
And this is the Error. ROM is 64Kx8

flandreas, 2022-02-02:
@leonuetzel Thank you! I'm still not able to reproduce the problem. On my test machines (Mac, Windows), I'm able to import your file without any problem.

The only thing I can imagine is that on machine the file lies at a location where Antares doesn't have enough privileges to read it. Although I suppose the file still lies at the location where Antares wrote it in the first place, so that would be strange as well.

Could you please repeat your test and then export the log file (using File -> Export Log File) and attach it here? I would expect to see some "No Access to File" exception in the log file.

leonuetzel, 2022-02-02:
[Antares-log.zip](https://github.com/flandreas/antares/files/7986407/Antares-log.zip)

flandreas, 2022-02-02:
Ok, it must be something else. I'm analyzing...

flandreas, 2022-02-02:
It seems like the problem is caused by memory contents popup window that issues an undoable command when importing or clearing contents, but doesn't correctly behave upon resulting data changes.

closed 2022-02-07 by flandreas