flandreas/antares #298: Flat Simulation Script Error

leonuetzel, 2022-02-02:
[image]
[image]
Tried to use Flat Simulation for a simple Multiplexer i created, but Antares throws Errors.
What am i doing wrong?

flandreas, 2022-02-02:
Your are trying to access individual bits of a bus input using an array accessor. That doesn't work.

You have to use the bit access operator @. Try the following:
```
OUT = IN@SW
```

flandreas, 2022-02-02:
BTW: For questions like this (which are always welcome), you can also use the project's discussion board.

leonuetzel, 2022-02-02:
Ah ok, thank you.
Next time i will use the discussion board.

flandreas, 2022-02-03:
Answered

closed 2022-02-03 by flandreas