flandreas/antares #301: Simulation and Editing large Circuits takes very long

leonuetzel, 2022-02-06:
I tried to simulate a quite big circuit and started the Simulation (Flat Simulation without showing Simulation Time) and waited for the Simulation to start. After 2 Hours it wasnt ready so i cancelled. CPU: i7-5500U, so not the slowest CPU...
Is it possible to optimize the Program so it simulates faster or maybe uses more Processor-Cores?

Similar Problem when drawing large Circuits: the Program freezes from time to time for like 5-10 secs if u edit something (maybe java garbagecollector?)

flandreas, 2022-02-06:
@leonuetzel Thank you for reporting this. Would you mind exporting and attaching your project so I can do some analysis?

leonuetzel, 2022-02-06:
[CPU.zip](https://github.com/flandreas/antares/files/8011044/CPU.zip)

flandreas, 2022-02-06:
Thank you. This is truly the largest Antares project I have ever seen. I'll use it to try to optimize various parts of Antares. Thanks again for sharing.

leonuetzel, 2022-02-06:
Yeah no problem. Try to edit some Things and you will see, that it freezes sometimes. I also noticed, that once a Simulation was loaded, restarting it was quite fast.

flandreas, 2022-02-07:
Performance at simulation start extracted to separate issue #302. With the current fix, I'm able to start the simulation of your CPU in about 10 secs on my machine.

flandreas, 2022-02-07:
@leonuetzel Simulation startup performance fixed with release 0.19.0. I couldn't reproduce the performance lags when editing your project, but your assumption (Java GC) could be right. I will keep track of this issue.

closed 2022-02-09 by flandreas