flandreas/antares #303: Opening ROM in Circuit results in Error

leonuetzel, 2022-02-07:
Trying to open
CPU->ProgramMemory->ROM
[image]

to edit the Memory results in an unexpected Error.

flandreas, 2022-02-07:
@leonuetzel Yeah, that's a bug. It occurs when you open a memory component in a circuit that is not the top-level circuit, i.e. after diving into a sub circuit. Never tried that before. Thank you for reporting this.

BTW: Editing ROM content in a sub circuit wouldn't be possible anyway. For editing, you would have to open "ProgramMemory" as top-level circuit.

leonuetzel, 2022-02-07:
Ah ok, good to know. Guess i will put the ROM Circuit directly into the CPU for easier testing.

flandreas, 2022-02-15:
Fixed with release 0.20.0

closed 2022-02-15 by flandreas