flandreas/antares #306: Drawing Wires

leonuetzel, 2022-02-09:
It would be nice to be able to create a new wire without the need to hold the mouse button pressed until you reach the target pin.
This would simplify the drawing of large circuits where you need to scroll to other areas of the circuit.

flandreas, 2022-02-10:
This is already implemented. Press ALT when starting to create a wire from a pin. The green connection indicator then changes from a circle to a rectangle. Press and release left mouse key to create the wire, and then you can move the mouse without holding a mouse button. Click the left mouse button to define intermediate points.

See also the user manual in https://www.antarescircuit.io/user-manual/english/circuits/wires.

leonuetzel, 2022-02-10:
Damn, sometimes i feel really stupid :/
Thank you

closed 2022-02-10 by leonuetzel