flandreas/antares #323: Preferences Crash

leonuetzel, 2022-03-12:
Going to "File"->"Settings"->"Circuit" and clicking the Arrow left to "Circuit" makes Antares infinitely crash. Only solvable with Taskmanager Shutdown.

flandreas, 2022-03-13:
@leonuetzel Thank you for reporting this. I can't reproduce the problem on any of my machines (Mac, Windows).

Could you please produce it once more, then kill Antares, and then grab your log file and attach it to this issue? If you are on Windows, you find the log file in ..\Users\YOUR_USER_NAME\AppData\Roaming\Antares\Antares.log

leonuetzel, 2022-03-13:
[Antares.log](https://github.com/flandreas/antares/files/8239330/Antares.log)

flandreas, 2022-03-13:
@leonuetzel Thanks for your quick response!

There is a text missing in the German translation files. I couldn't reproduce it because I've tested with English.

If you want to explore the new preference, you could workaround the problem by temporarily changing your language to English, explore/change the new preference, and then switch back to German.

flandreas, 2022-03-13:
Developer note: Missing german translation for antares.preferences.group.expression.

leonuetzel, 2022-03-13:
Ok, thank you.

flandreas, 2022-03-16:
Released with 0.22.0

closed 2022-03-16 by flandreas