flandreas/antares #341: Bus Transceiver - Strange Behaviour

Kusselkopp, 2022-04-04:
I want to implementing a registerfile with dedicated data input and data output ports.
However, I have the problem that the bus transceiver have sometimes a problem to get into a stable state.
A week (or two) ago, I always got the problem that they never stopped, i.e., setting B/A to 0, then a value, then 0, then a value, etc. for a very long time until I finally stopped simulation. It looked like a live lock.

Today, with the newest Antares version I gave it another try and had problems recreating this problem but eventually I managed to get a similar behaviour while recording it.

Here are the links:
https://www.youtube.com/watch?v=I2yN2jEcvC8
https://www.youtube.com/watch?v=fQAG3Eu_WRk

flandreas, 2022-04-04:
@Kusselkopp Thank your for reporting. I will look into this.

flandreas, 2022-04-04:
@Kusselkopp Definitively a bug. I can reproduce the problem with the minimal circuit of a single 4-Bit-Transceiver. As soon as the !EN input is set to 1, the circuit starts to oscillate. We haven't yet unit test for this scenario, so we will setup one and try to fix the problem with high priority.

flandreas, 2022-04-05:
@Kusselkopp I've fixed the bug and built a 0.24.1-beta release (not yet publicly available). I've also extended my unit test suite for these scenarios, and they all pass, but may I ask you to do some testing before I release it? You can download the beta release from https://github.com/flandreas/antares/releases/tag/v0.24.1-beta.

BTW: I had to slightly change the circuit to do some manual R/W testing , e.g. inverted the EN signals of the  transceiver to match WR of the RAM:

[image]

Kusselkopp, 2022-04-05:
@flandreas I did some tests with v0.24.1-beta and I was not able to recreate the problem - so everything works now.
Also the performance seemed to have improved.

Yes one circuit was a bit flawed. :)

Thank you very much.

flandreas, 2022-04-05:
@Kusselkopp Great, thank you for your testing!

flandreas, 2022-04-06:
Released in 0.25.0

closed 2022-04-06 by flandreas