flandreas/antares #348: Bit Extender doesn't save "Output bit width"

XBlayz, 2022-04-09:
Bit Extender does not save the "Output bit width" parameter, every time I reload the circuit it resets to 1.

flandreas, 2022-04-09:
@XBlayz Thank you for reporting this.

flandreas, 2022-04-09:
@XBlayz The bug is in the writing logic. We will fix that as soon as possible. In the meantime, as a workaround, you could manually change the saved file so that work on your project is not blocked.

Go to Antares' data directory on your computer. On Mac, this is /Users/{Username}/Library/Application Support/Antares/projects. Open dictionary.xml to look up the UUID of your projects. Go to the corresponding subdirectory and lookup in directory.xml the UUID of your circuit. Open the corresponding circuit file <UUID>.cir and manually edit the `outputBitWith` property to a valid bit width number.

Example:

```
<metaGraph _id='0'>
  <graph>
    <graphStorable _id='1'>
      <model>
        <graph _id='2'>
          <elements>
            <bitExtender _id='3' id='1' delay='20' inputBitWidth='1' outputBitWidth='16'/>
          </elements>
        </graph>
      </model>
```

Unfortunately, you will have to repeat this manual fix every time you've saved your circuit.

XBlayz, 2022-04-09:
Thanks for your help.

flandreas, 2022-04-21:
Released with 0.26.0

closed 2022-04-21 by flandreas