flandreas/antares #354: Renaming Circuits doesnt update already placed Circuits Names

leonuetzel, 2022-04-24:
Renaming a Circuit that is a Subcircuit in another Circuit, doesnt update that Subcircuits Name.

flandreas, 2022-04-26:
@leonuetzel Thank you for reporting this.

I'm assuming you are referring to the following situation:
- You have a circuit A
- You have a circuit B that uses A
- You open B
- You select A in the explorer tree and change its name to "AA" using popup action "Rename.."
- You'd expect a tooltip saying "AA" if you hover over the symbol of AA in the still open circuit B

Then you are right; the tooltip still says "A". If close and reopen B, then the tooltip correctly says "AA". So changed names are correctly applied if the using circuit is reopened, but they are NOT correctly applied WHILE the using circuit remains open.

leonuetzel, 2022-04-26:
Correct.