flandreas/antares #369: Ctrl+Z sometimes changes Things that it shouldnt

leonuetzel, 2022-05-05:
Sometimes Ctrl+Z does Things additionally to the Undo-Action, that ive never done before. It occurs unpredictably and is not reproducible. Often it does Things like changing the Position of Wires or Subcircuits.

flandreas, 2022-05-05:
@leonuetzel Thank you for reporting this.

May I ask to do the following to help me find the cause of the problem?

1. Make sure Antares logs your user actions by going to Preferences -> General and making sure that log level "Debug" is active.
2. The next time you observe the problem:
2.1 Export the log file using File -> "Export Logfile" and attach the ZIP to this issue
2.2 Describe the wrong change in your words, such as "AND gate with ID 25 changes position". You can determine the ID of the component by selecting it and inspecting property "ID" in the properties window.
2.3 Ideally, also export your project and attach it to this issue, so that I can replay your actions under the microscope

leonuetzel, 2022-05-05:
Ok, i enabled Debug-Loblevel. Waiting now for it to happen again :)

flandreas, 2022-07-16:
@leonuetzel I found the cause of the bug during my own work. Severe bug in the undo/redo system. Will fix as soon as possible.

closed 2022-07-21 by flandreas